hankhank10/music-screen-api #4: Artist and Album line breaks, too

Shadester, 2020-06-20:
Looks like a lot of times album names and artists can be too long, too. It would be great to have reflow on that too.

hankhank10, 2020-06-20:
It should line break automatically. Can you send me a photo of an example where it's not working right please?

Shadester, 2020-06-20:
[image]
This one is an example of both Album and Artist

hankhank10, 2020-06-20:
Ok thanks. I’ll have a look into it. Line breaks are actually kind of a pain so I just did it for the title but if it’s proving an issue then I can try to incorporate for the other two lines as well.

By the way I just pushed an update which significantly improves the appearance as it offers a white on black mode which looks pretty cool:

[image]

You can update using git pull - please do let me know if you run into any issues.

Shadester, 2020-06-20:
Thanks! Yeah, it would be sweet to have it wrap for the other ones too. I guess if extracting the wrapping into a function, it might be easier.

I have also been toying with the idea of using https://github.com/torfsen/python-systemd-tutorial to avoid having to use pm2 (since installing npm and pm2 installs tons of packages).

hankhank10, 2020-06-21:
Thanks. Yeah you should be able to start go_sonos with systemd or supervisor if you prefer and no changes to the code should be necessary.

I just really like pm2 because you can do everything from the command line rather than messing around with config files, I find the status output easier to see what’s going on with and I really like the web dashboard they have for remote monitoring. But I recognise it’s not that lightweight.

closed 2020-07-08 by hankhank10