hankhank10/music-screen-api #16: Album Art not displaying - NAS

rdrivas123, 2020-07-14:
When using music from a local NAS, the album artwork does not display on the RasPi, it only shows the Sonos logo.
The artwork shows correctly in the Sonos IOS app as well as on the desktop PC app.

rdrivas123, 2020-07-16:
I tried to run it with dev, but the app wont start, I end up having to do a fresh install of the entire Pi to get it back to the master version. Whenever I try to use Dev it just wont work at all.

jjlawren, 2020-07-16:
That shouldn't be necessary. It sounds like you're having issues with `git`. Can you share the output that's preventing you from checking out `master`?

What exactly doesn't work in the `dev` branch?

rdrivas123, 2020-07-16:
I saw your message on Reddit, I should probably be clear that i am no expert when it comes to Pi or code, so I am surely not doing something right, I have the master running again, the artwork still doesn't show up from the NAS< but does work from Amazon or Sonos Radio.

I can do a screen share if you are feeling like being that helpful so you can see the Pi and possibly make adjustments to the dev code?

jjlawren, 2020-07-16:
The way the config is loaded now seems to be causing problems during upgrades or switching branches. I'm working on a change that may require a one-time extra step during the next upgrade, but should make it more stable moving forward.

If you're still having issues after that I can suggest some other options.

jjlawren, 2020-07-18:
@rdrivas123 can you try to update to the latest `dev` with the following steps:
1. Move the `sonos_settings.py` file to a backup location, like `sonos_settings.py.bak`
2. Run `git pull`
3. If you're still on `master`, run `git checkout dev`
4. Copy your backed-up config file to `sonos_settings.py`
5. Add this line to the settings file if it doesn't exist already:
```
log_file = "~/music-screen-api.log"
```
6. Start up and share the log from `/home/pi/music-screen-api.log` when you see the album art fail to load.

rdrivas123, 2020-07-19:
Tried that and I get an error:
I put that line in the general settings area in the sonos_settings.py file.

  File "music-screen-api/go_sonos_highres.py", line 21, in <module>
    import sonos_settings
  File "/home/pi/music-screen-api/sonos_settings.py", line 7
    log_file - +~/music-screen-api.log"

rdrivas123, 2020-07-19:
Got it to start, I had some typos in the line, but now all i see is a black screen and no album art from anything

jjlawren, 2020-07-19:
Anything logged?

Are you sure `node-sonos-http-api` started properly? Can you connect to `http://<IP_OF_PI>:5005` in your browser?

rdrivas123, 2020-07-19:
[image]

I can connect, still no artwork at all.

jjlawren, 2020-07-19:
Sorry, I'll need some type of output from the script to make any guesses. If there's nothing written to the log file, you can also check `~/.xsession-errors` if you've set it to autostart like in the installation instructions. This might tell you if anything is going wrong at boot.

If you're relying on restarts of the Pi to test your changes, try this to make testing a bit easier:
1. SSH into the Pi.
2. Stop the running script with `pkill python`.
3. Run the script manually with: `DISPLAY=:0 /usr/bin/python3 /home/pi/music-screen-api/go_sonos_highres.py`. The script should output directly to the screen.
4. Stop the script by pressing Control-C.
5. Repeat from step 3 as needed to test more changes.

rdrivas123, 2020-07-20:
Did a reboot and now artwork is showing up from the NAS device as expected.
I did get the latest dev as you instructed, not sure if there was a code change to fix it in there.

jjlawren, 2020-07-20:
Good to hear. There were some improvements in dev compared to master.

Is everything working as you expect now? Should we close this?

rdrivas123, 2020-07-20:
Yes, working as expected, thanks for the work on this and the overall project!

closed 2020-07-20 by rdrivas123