hankhank10/music-screen-api #19: rpi-backlight not disabling backlight

FerretVW, 2020-07-17:
I have a fresh install on a raspberry pi 3B+, this is the only thing running on it, I tried adding rpi-backlight to have the backlight dim when music is not playing but it does not seem to work (ran the installer as well as the permission modify). Other than the install of this I only have 2 modifications done to /boot/config.txt; one to rotate the screen 180 (rotate_display=2) and one to turn off the backlight when the pi is off (dtoverlay=gpio-poweroff,gpiopin=19,active_low=1). Any additional documentation for this to work?

jjlawren, 2020-07-18:
Did you reboot after setting the udev rules? Check the permissions on your backlight power control. It should look like this:
```
$ ls -l /sys/class/backlight/*/bl_power
-rw-rw-rw- 1 root root 4096 Jul 17 21:00 /sys/class/backlight/rpi_backlight/bl_power
```
What does your `/etc/xdg/lxsession/LXDE-pi/autostart` file look like? I've found the `sudo` prefix is unnecessary (and possibly a security risk). For some reason the root user can't always load python modules installed via pip without some environment variable changes. The script startup line should work fine set like this:
```
@/usr/bin/python3 /home/pi/music-screen-api/go_sonos_highres.py
```
Finally, if none of that works, make sure you have this line in your `sonos_settings.py` file:
```
log_file = "~/music-screen-api.log"
```
That will write a file to the log location above which will help troubleshoot the issue.

FerretVW, 2020-07-18:
I have rebooted after applying the rule. When I try to check permissions I get:
`ls: cannot access '/sys/class/backlight/*/bl_power': No such file or directory`
I think this is related to this when I try to start go_sonos_highres.py manually:
`FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/backlight/rpi_backlight/max_brightness'`
sudo was set on that line, removed it and rebooted and now go_sonos_highres.py doesn't start, re-added sudo and it starts on boot again. Logging is enabled now. For completeness, here is the output of pip3 install rpi-backlight:
`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: rpi-backlight in ./.local/lib/python3.7/site-packages (2.2.0)`
as well as the permissions modification:
`echo 'SUBSYSTEM=="backlight",RUN+="/bin/chmod 666 /sys/class/backlight/%k/brightness /sys/class/backlight/%k/bl_power"' | sudo tee -a /etc/udev/rules.d/backlight-permissions.rules
SUBSYSTEM=="backlight",RUN+="/bin/chmod 666 /sys/class/backlight/%k/brightness /sys/class/backlight/%k/bl_power"`

jjlawren, 2020-07-18:
Are you using the HyperPixel display or a different one?

It looks like you've installed `rpi-backlight` in a local directory for the pi user. Try to install for all users (including root) with:
```
sudo pip3 install rpi-backlight
```

FerretVW, 2020-07-18:
Hyperpixel, non-touch. Installing rpi-backlight with that command yielded no change for the backlight and stopped go_sonos_highres.py from running at boot.

jjlawren, 2020-07-18:
Check your `/home/pi/.xsession-errors` file for errors encountered during auto start.

Can you try to run it manually? Over ssh you can make it run on the screen with:
```
$ DISPLAY=:0 /usr/bin/python3 /home/pi/music-screen-api/go_sonos_highres.py
```
Share anything that's logged if that still has issues.

FerretVW, 2020-07-18:
contents of /home/pi/.xsession-errors, nothing jumping out except the unclutter error at the end:
`Xsession: X session started for pi at Fri 17 Jul 2020 10:09:00 PM CDT
dbus-update-activation-environment: setting DBUS_SESSION_BUS_ADDRESS=unix:path=$
dbus-update-activation-environment: setting DISPLAY=:0
dbus-update-activation-environment: setting XAUTHORITY=/home/pi/.Xauthority
localuser:pi being added to access control list
dbus-update-activation-environment: setting USER=pi
dbus-update-activation-environment: setting LANGUAGE=en_US.UTF-8
dbus-update-activation-environment: setting TEXTDOMAIN=Linux-PAM
dbus-update-activation-environment: setting XDG_SESSION_TYPE=x11
dbus-update-activation-environment: setting HOME=/home/pi
dbus-update-activation-environment: setting DESKTOP_SESSION=lightdm-xsession
dbus-update-activation-environment: setting XDG_SEAT_PATH=/org/freedesktop/Disp$
dbus-update-activation-environment: setting DBUS_SESSION_BUS_ADDRESS=unix:path=$
dbus-update-activation-environment: setting LOGNAME=pi
dbus-update-activation-environment: setting XDG_SESSION_CLASS=user
dbus-update-activation-environment: setting PATH=/home/pi/.local/bin:/usr/local$
dbus-update-activation-environment: setting XDG_SESSION_PATH=/org/freedesktop/D$
dbus-update-activation-environment: setting XDG_RUNTIME_DIR=/run/user/1000
dbus-update-activation-environment: setting DISPLAY=:0
dbus-update-activation-environment: setting LANG=en_US.UTF-8
dbus-update-activation-environment: setting XDG_SESSION_DESKTOP=lightdm-xsession
dbus-update-activation-environment: setting XAUTHORITY=/home/pi/.Xauthority
dbus-update-activation-environment: setting XDG_GREETER_DATA_DIR=/var/lib/light$
dbus-update-activation-environment: setting SHELL=/bin/bash
dbus-update-activation-environment: setting GDMSESSION=lightdm-xsession
dbus-update-activation-environment: setting NO_AT_BRIDGE=1
dbus-update-activation-environment: setting GPG_AGENT_INFO=/run/user/1000/gnupg$
dbus-update-activation-environment: setting LC_ALL=en_US.UTF-8
dbus-update-activation-environment: setting PWD=/home/pi
** Message: 22:09:10.185: main.vala:101: Session is LXDE-pi
** Message: 22:09:10.185: main.vala:102: DE is LXDE
** Message: 22:09:10.611: main.vala:133: log directory: /home/pi/.cache/lxsessi$
** Message: 22:09:10.611: main.vala:134: log path: /home/pi/.cache/lxsession/LX$
/usr/bin/unclutter: someone created a sub-window to my sub-window! giving up`

When running via SSH I appear to be getting the same results as when I tried to start it manually inside VNC:
`Traceback (most recent call last):
  File "/home/pi/music-screen-api/go_sonos_highres.py", line 24, in <module>
    backlight = Backlight()
  File "/home/pi/.local/lib/python3.7/site-packages/rpi_backlight/__init__.py", line 71, in __init__
    self._max_brightness = self._get_value("max_brightness")  # 255
  File "/home/pi/.local/lib/python3.7/site-packages/rpi_backlight/__init__.py", line 85, in _get_value
    raise e
  File "/home/pi/.local/lib/python3.7/site-packages/rpi_backlight/__init__.py", line 77, in _get_value
    return int((self._backlight_sysfs_path / name).read_text())
  File "/usr/lib/python3.7/pathlib.py", line 1199, in read_text
    with self.open(mode='r', encoding=encoding, errors=errors) as f:
  File "/usr/lib/python3.7/pathlib.py", line 1186, in open
    opener=self._opener)
  File "/usr/lib/python3.7/pathlib.py", line 1039, in _opener
    return self._accessor.open(self, flags, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/backlight/rpi_backlight/max_brightness'`

jjlawren, 2020-07-18:
I'm not sure why you're missing (or can't access) the `max_brightness` file. Here's what my permissions look like:
```
$ ls -l /sys/class/backlight/rpi_backlight/
total 0
-r--r--r-- 1 root root 4096 Jul 17 22:19 actual_brightness
-rw-rw-rw- 1 root root 4096 Jul 17 22:19 bl_power
-rw-rw-rw- 1 root root 4096 Jul 13 18:06 brightness
lrwxrwxrwx 1 root root    0 Jul 17 22:19 device -> ../../../rpi_backlight
-r--r--r-- 1 root root 4096 Jul 13 18:06 max_brightness
drwxr-xr-x 2 root root    0 Jul 17 22:19 power
lrwxrwxrwx 1 root root    0 Jul 13 18:06 subsystem -> ../../../../../class/backlight
-r--r--r-- 1 root root 4096 Jul 13 18:06 type
-rw-r--r-- 1 root root 4096 Jul 13 18:06 uevent
```
I'm running the same non-touch display on a Pi 4.

FerretVW, 2020-07-18:
Yeah, I am baffled by this also, the whole /sys/class/backlight/ directory is just empty, I am seeing others having the same issue with other screens/brightness issues, but this is a fresh install of raspbian (and the third fresh install at that)

jjlawren, 2020-07-18:
Did you install the desktop or lite version of Raspbian? A recent release?

FerretVW, 2020-07-18:
Desktop, 5/27/2020 release (Default top option when imaging with Raspberry Pi Imager v1.3). When I start from scratch again tomorrow maybe I'll try full just in case stuff is pulled from the recommended img.

jjlawren, 2020-07-18:
Odd. I used the plain desktop version and didn't have to jump through any hoops to get the backlight to work besides set the udev rules.

How did you setup the display? Did you use the pimoroni installer from [https://github.com/pimoroni/hyperpixel4](https://github.com/pimoroni/hyperpixel4)?

FerretVW, 2020-07-18:
Yeah, used these exact steps:
For Raspbian install (used this in case his setup method was any different than my own): https://www.hackster.io/mark-hank/super-simple-headless-raspberry-pi-setup-5382d6

And this for the setup for display:
https://www.hackster.io/mark-hank/sonos-album-art-on-raspberry-pi-screen-5b0012

jjlawren, 2020-07-18:
I'm running low on ideas. Does anything get returned when running:
```
sudo find /sys/ -type f -iname '*brightness*'
```
From https://askubuntu.com/a/715310.

FerretVW, 2020-07-18:
OK, good news, re-installed this AM and all is working, I did the install in a different order though (installed rpi-backlight after the sonos-http-api and before the music-screen-api)and used the dev build rather than master. everything else stayed identical. My guess is I wasn't backing out of the directory or something before installing rpi-backlight. Thank you so much.

closed 2020-07-18 by FerretVW

jjlawren, 2020-07-18:
Great! Now that you're on `dev` let me know if you run into any issues. It's been pretty stable for me, but good to have confirmation from others.

FerretVW, 2020-07-18:
Only thing not working is enabling webhook support, I am sure that is user error though, added:
```
{
  "webhook": "http://localhost:8080/"
}
```
to the end of settings.js in node-settings-http-api but just got a black screen on launch then, so reverted. Truly not worried about it though, the constant pole was bothersome when the screen was bare but in the enclosure that bottom .5mm of the screen is hidden so I don't see it

jjlawren, 2020-07-18:
Ah yeah that's confusing. Don't touch the `settings.js` file as that's part of the `node-sonos-http-api` code. You should put it in a file named `settings.json` in that same directory. It probably doesn't exist yet. I'll update the docs to make this more clear as others have run into this same confusing step.

FerretVW, 2020-07-18:
perfect, thank you so much. Everything is working as it should now.