hankhank10/music-screen-api #20: Sirius Not Refreshing

rebrook, 2020-07-24:
Thank you for this! I am running into an issue where the art refreshes as expected when playing Apple Music and when I start a station from Sirius. However, when a song changes on Sirius it doesn't update to the next song (it remains on the first song played on the station). If I change stations, the artwork updates as expected, but not when the next song plays.

Have you encountered this before and are there any known fixes?

I am on the dev branch, but it was also happening on the main branch.

jjlawren, 2020-07-24:
Can you do a `git pull` to get the latest changes?

Do you see messages about `New track:` in the logs when the song changes on Sirius?

rebrook, 2020-07-24:
yes, on the latest changes. Below is log where first song works and then nothing happens after song switches.

2020-07-24 11:32:10,063    INFO - Writing to log file: /home/pi/music-screen-api.log
2020-07-24 11:32:10,065    INFO - Monitoring room: Office
2020-07-24 11:32:10,088   DEBUG - New track: 12 - PopRocks - 12 - PopRocks () - 0:00:00
Just using offline as we are in offline_only_mode
2020-07-24 11:32:10,090   DEBUG - Demastered to 12 - PopRocks
2020-07-24 11:32:52,142    INFO - Switching to webhook updates
^C2020-07-24 11:36:44,437   DEBUG - Shutting down

Thanks!

jjlawren, 2020-07-24:
Hmm, when you look in the Sonos app does it display title/artist/etc for the currently playing track? Or just the name of the radio station while the album art changes?

Getting output from the Sonos API would help confirm this. Can you send the output from these two commands both before and after the song changes? I'm assuming you'll run them from on the Pi itself over SSH.
```
curl http://localhost:5005/Office/state
curl http://localhost:8080/status
```

rebrook, 2020-07-24:
In the Sonos App, it takes about 30 seconds to switch the album art to the next song.

Before:

{"volume":9,"mute":false,"equalizer":{"bass":0,"treble":0,"loudness":true},"currentTrack":{"artist":"16 - The Blend","title":"BR P|TYPE=SNG|TITLE Style|ARTIST Taylor Swift|ALBUM","albumArtUri":"http://albumart.siriusxm.com/albumart/1700/NDCA-000285381-001_m.jpg","duration":0,"uri":"x-sonosapi-hls:r%3astarlite?sid=37&flags=8480&sn=2","trackUri":"x-sonosapi-hls:r%3astarlite?sid=37&flags=8480&sn=2","type":"radio","stationName":"16 - The Blend","absoluteAlbumArtUri":"http://albumart.siriusxm.com/albumart/1700/NDCA-000285381-001_m.jpg"},"nextTrack":{"artist":"","title":"","album":"","albumArtUri":"","duration":0,"uri":""},"trackNo":1,"elapsedTime":6,"elapsedTimeFormatted":"00:00:06","playbackState":"PLAYING","playMode":{"repeat":"none","shuffle":false,"crossfade":false}}

{"room": "Office", "status": "PLAYING", "trackname": "16 - The Blend", "artist": "16 - The Blend", "album": "", "duration": 0, "last_poll": 1595611390.2315552, "last_webhook": 1595611456.0021896, "webhook_active": true}

After:

{"volume":9,"mute":false,"equalizer":{"bass":0,"treble":0,"loudness":true},"currentTrack":{"artist":"16 - The Blend","title":"BR P|TYPE=SNG|TITLE Big Yellow Taxi|ARTIST Counting Crows|ALBUM","albumArtUri":"http://albumart.siriusxm.com/albumart/1700/NDCA-000103534-002_m.jpg","duration":0,"uri":"x-sonosapi-hls:r%3astarlite?sid=37&flags=8480&sn=2","trackUri":"x-sonosapi-hls:r%3astarlite?sid=37&flags=8480&sn=2","type":"radio","stationName":"16 - The Blend","absoluteAlbumArtUri":"http://albumart.siriusxm.com/albumart/1700/NDCA-000103534-002_m.jpg"},"nextTrack":{"artist":"","title":"","album":"","albumArtUri":"","duration":0,"uri":""},"trackNo":1,"elapsedTime":105,"elapsedTimeFormatted":"00:01:45","playbackState":"PLAYING","playMode":{"repeat":"none","shuffle":false,"crossfade":false}}

{"room": "Office", "status": "PLAYING", "trackname": "16 - The Blend", "artist": "16 - The Blend", "album": "", "duration": 0, "last_poll": 1595611390.2315552, "last_webhook": 1595611555.411524, "webhook_active": true}

jjlawren, 2020-07-24:
Okay, I see the issue. The "title" value is changing, but this field is currently ignored for radio stations.

I was already reworking this part so I'll ping you when a new version is ready to test.

jjlawren, 2020-07-24:
A partial fix is available now. I have a better solution in mind for radio stations, but it's not ready yet. Can you let me know if it behaves any better with Sirius after pulling the latest commits?

rebrook, 2020-07-24:
Awesome! Works as expected now. Thanks again!

closed 2020-07-24 by jjlawren