hankhank10/music-screen-api #21: "No module named 'aiohttp'"

habra-cadabra, 2020-07-24:
Hello!

was having the same issues covered in #16 & #19 so I decided to switch to the dev branch temporarily. when attempting to start go_sonos_highres.py up to test I get the following error:
```
$ DISPLAY=:0 /usr/bin/python3 /home/pi/music-screen-api/go_sonos_highres.py
Traceback (most recent call last):
  File "/home/pi/music-screen-api/go_sonos_highres.py", line 16, in <module>
    from aiohttp import ClientError, ClientSession
ModuleNotFoundError: No module named 'aiohttp'
```
switched back to the origin branch and everything is back to normal. it's very likely I'm doing something wrong, but I can't locate this file anywhere. 

Thanks!

jjlawren, 2020-07-24:
It's a new dependency. Just run this:
```
sudo pip3 install aiohttp
```

habra-cadabra, 2020-07-24:
that was it, thank you for your help.

closed 2020-07-24 by jjlawren