hankhank10/music-screen-api #22: Album Art showing Playlist Artwork not Track Artwork

craigusus, 2020-07-25:
Hi,

Firstly, thanks for this and for [vinylemulator](https://github.com/hankhank10/vinylemulator).

Got this up and running pretty quick, even on a Pi Zero W and the rectangular [Hyperpixel 4.0](https://shop.pimoroni.com/products/hyperpixel-4)!

Is there a way to have the display show the cover art for the track that is playing from the play list rather than the cover art for the whole playlist?
ie this is the result of http://192.168.1.81:5005/Lounge/state
```
{
  "volume": 34,
  "mute": false,
  "equalizer": {
    "bass": 0,
    "treble": 0,
    "loudness": false
  },
  "currentTrack": {
    "artist": "The National Parks",
    "title": "As We Ran",
    "album": "As We Ran",
    "albumArtUri": "/getaa?s=1&u=x-sonos-spotify%3aspotify%253atrack%253a5ITmuvjtUUdxVQCCsaKBDx%3fsid%3d9%26flags%3d8224%26sn%3d3",
    "duration": 200,
    "uri": "x-sonos-spotify:spotify%3atrack%3a5ITmuvjtUUdxVQCCsaKBDx?sid=9&flags=8224&sn=3",
    "trackUri": "x-sonos-spotify:spotify%3atrack%3a5ITmuvjtUUdxVQCCsaKBDx?sid=9&flags=8224&sn=3",
    "type": "track",
    "stationName": "",
    "absoluteAlbumArtUri": "https://seeded-session-images.scdn.co/v1/img/artist/6a1PrMzNmhNE8HrO8burHc/en"
  },
  "nextTrack": {
    "artist": "Ivan & Alyosha",
    "title": "The Worth Of The Wait",
    "album": "The Worth Of The Wait",
    "albumArtUri": "/getaa?s=1&u=x-sonos-spotify%3aspotify%253atrack%253a4ngv7whSPSAwD8ZP3zMHsd%3fsid%3d9%26flags%3d8224%26sn%3d3",
    "duration": 171,
    "uri": "x-sonos-spotify:spotify%3atrack%3a4ngv7whSPSAwD8ZP3zMHsd?sid=9&flags=8224&sn=3",
    "trackUri": "x-sonos-spotify:spotify%3atrack%3a4ngv7whSPSAwD8ZP3zMHsd?sid=9&flags=8224&sn=3",
    "absoluteAlbumArtUri": "http://192.168.1.80:1400/getaa?s=1&u=x-sonos-spotify%3aspotify%253atrack%253a4ngv7whSPSAwD8ZP3zMHsd%3fsid%3d9%26flags%3d8224%26sn%3d3"
  },
  "trackNo": 37,
  "elapsedTime": 86,
  "elapsedTimeFormatted": "00:01:26",
  "playbackState": "PLAYING",
  "playMode": {
    "repeat": "none",
    "shuffle": false,
    "crossfade": false
  }
}
```

The display seems to be showing the artwork from
```
"absoluteAlbumArtUri": "https://seeded-session-images.scdn.co/v1/img/artist/6a1PrMzNmhNE8HrO8burHc/en"
```
Rather than:
```
"albumArtUri": "/getaa?s=1&u=x-sonos-spotify%3aspotify%253atrack%253a5ITmuvjtUUdxVQCCsaKBDx%3fsid%3d9%26flags%3d8224%26sn%3d3"
```

Playlist URI for reference: spotify:playlist:37i9dQZF1E4jT7CxMP1UY4

jjlawren, 2020-07-25:
This should be fixed in the `dev` branch. Can you give that a shot?

craigusus, 2020-07-26:
@jjlawren 
Great, that works a treat, thanks!
I did a bit of tweaking in ```go_sonos_highres.py``` 
```
thumbsize = 400,400
#thumbsize = 600,600   # pixel size of thumbnail if you're displaying detail
screensize = 800,480
#screensize = 720,720  # pixel size of HyperPixel 4.0

...

root.geometry("800x480")
#root.geometry("720x720")
root.title("Music Display")

album_frame = tk.Frame(root, bg='black', width=800, height=480)
curtain_frame = tk.Frame(root, bg='black', width=800, height=480)
#album_frame = tk.Frame(root, bg='black', width=720, height=720)
#curtain_frame = tk.Frame(root, bg='black', width=720, height=720)
```

Also tweaked the font sizes to be a little smaller. Artwork isnt quite centered in landscape, but I'm more than happy with that.

[image]

I did notice that when I came back to it this morning, after stopping Sonos from the desktop app and leaving it for about 8 hours, it wouldn't start again. No response from keyboard or mouse either. I had to reboot the Pi by pulling the power out. Is there a power setting I could adjust so it can stay 'active' 24/7 ?

jjlawren, 2020-07-26:
Thanks for testing on the rectangular one. I have an idea of how to add better support for that size natively soon.

If you're running on a Zero, make sure you've set up the webhook support as shown in the `dev` README. It will reduce the strain on the underpowered processor.

You shouldn't need to do anything to keep it available after long periods of inactivity. If it happens again check the logs for both this script and `node-sonos-http-api` and see if anything stands out.

craigusus, 2020-08-01:
hi @jjlawren,

Thanks for the quick response.

Still unsure as to why it was not coming back on. Nothing in the logs for this, and couldn't find where the logs were for ```node-sonos-http-api```. Other than it loosing network connection so not seeing the Sonos controller, meaning the display staying off, so not being able to see any keyboard/mouse input. Could ```hyperpixel_backlight.py``` allow the backlight to come on from local keyboard/mouse input?

I've now 'upgraded' the Pi Zero to a Pi 2B plus a USB WiFi dongle and seems to be more reliable.

jjlawren, 2020-08-01:
The logs will say when it changes tracks (and updates the display). If you're seeing this then perhaps review the setup guide which explains how to turn off the display power saving and screensaver. With backlight control in the script we handle that ourselves now.

craigusus, 2020-08-03:
I think this is down to either my WiFi or the USB dongle and loosing connection somewhere. Has been idle for 2 days wired via Ethernet and worked straight away from playing something from the controller. New dongle ordered. Closing.

closed 2020-08-03 by craigusus