hankhank10/music-screen-api #23: image quality on latest dev branch is poor

habra-cadabra, 2020-07-28:
hello!

updated to the latest dev build today and I noticed the image quality is poor when comparing to earlier versions. I rolled back the to the 799a2f2 build and the image looked better. 

the source isn't great in the example images, as Apple Music's supplied artwork for this album is only 400x400 pixels,
but you can see a difference between the two. I also noticed the default sonos.png file looked pixelated on the latest build. 

latest build:
[image]

799a2f2 build
[image]

(one final note, the only change I made to the file was to change the font size of the track title from 40 to 30)

jjlawren, 2020-07-28:
Ah, I see why. I misread some library documentation and mistakenly switched to a lower quality resizing algorithm. Easy fix.

jjlawren, 2020-07-28:
Should be fixed in ed1a358a8170a13dbb1598e1ec579f662f4c3657, let me know if it looks better.

habra-cadabra, 2020-07-28:
look better! 

thanks so much for your continued work on this project, it's very cool.

closed 2020-07-28 by jjlawren