hankhank10/music-screen-api #25: Setup for 2 Pi's working together?

yankeeatz, 2020-07-29:
I suppose this is more of a question than an issue.  I have a Pi Zero on the way that I plan to connect to the Inky screen.  From reading previous write-ups, I know I can also use a Pi 3 to run the server since it has more horsepower than the Zero.

Which section of the instructions do I need to run on the Pi 3?  My assumption was I would only need to install node-sonos-http-api (server) but when I skipped the install of music-screen-api, I couldn't get the server to run.  It also seems like the server does not start after reboot.

I think I have the Zero side figured out - everything gets installed, and then I just edit the config files to point it to the Pi 3 to pull the info from the server.  

Thanks!

craigusus, 2020-08-01:
Hi, @yankeeatz 

I'm looking at doing something similar as I've also got [vinylemulator](https://github.com/hankhank10/vinylemulator) running on a seperate Pi.

I would go with the below, not tested though...
On your Pi 3 from these [instrutions](https://www.hackster.io/mark-hank/currently-playing-music-on-e-ink-display-310645):
Follow **Step 1: Set up your Raspberry Pi**
Follow **Step 3: Raspberry Pi housekeeping**
Follow **Step 5: Install node-sonos-http-api (assuming you want to use Sonos)**
Follow **Step 7: Make everything run on startup and all the time** - but ignore the lines:
```Now let's get the music-screen-api to run on startup too.```
to
```It should show you a status screen which shows go_sonos and npm (if you're running Sonos) both as running.```

Then in ```sonos_settings.py``` on your Pi Zero you'll need to edit the line ```sonos_http_api_address = "localhost"``` to ```sonos_http_api_address = "your Pi 3's IP address"```. Then reboot. But I think you've already sorted this.

hankhank10, 2020-08-01:
This is correct. Get the Sonos API running on the Pi 3, then connect to it from the Pi running the screen by changing the IP address from localhost to the IP of the Pi 3.

yankeeatz, 2020-08-01:
Thanks, working on the Pi3 setup now.  Question about some of the things in Step 7.    Are the instructions starting with "Assuming pm2 is all good let's tell pm2 to run itself on start" needed?  I haven't done that, but node-sonos-http-api server is starting after every reboot as it should.

yankeeatz, 2020-08-03:
OK, first actual issue.  The Pi3 seems to be running just fine.  Setting up the Zero now.  I have gotten to the install of the Inky library, which worked fine.  Next step gives an issue:
git clone https://github.com/hankhank10/music-screen-api.git

gives the output:
-bash: git: command not found

and music-screen-api won't install.  It's almost like the Inky install "broke" git or something?  If it matters, I have using Raspberry Pi Lite on the Zero, running all of this over SSH.

craigusus, 2020-08-03:
@yankeeatz git is not pre installed on the Raspberry OS Lite.
You'll have to run `sudo apt install git` to install before running the got clone command.

yankeeatz, 2020-08-03:
Thanks, I was just updating my comment, to say I had figured that out.  Didn't realize it wasn't pre-installed but did find the command to install it.  On to the next step!

yankeeatz, 2020-08-03:
Another issue!  I thought it might be caused (again) by something not installed with the Lite OS so redid everything with the Desktop/GUI version and same error.  There go a few hours wasted!

pi@raspberrypizero:~ $ python3 music-screen-api/go_sonos.py
Traceback (most recent call last):
  File "music-screen-api/go_sonos.py", line 7, in <module>
    import ink_printer # does the printing to ink
  File "/home/pi/music-screen-api/ink_printer.py", line 3, in <module>
    from font_source_serif_pro import SourceSerifProSemibold
ModuleNotFoundError: No module named 'font_source_serif_pro'

Thanks to everyone for the support, I did not expect to have this many problems getting this running!

closed 2020-08-03 by yankeeatz