hankhank10/music-screen-api #26: Support for rectangle HyperPixel display

jjlawren, 2020-07-30:
> @rdrivas123 I have an idea of how to make this work on both HyperPixel screen dimensions but haven't had time to play with it as of yet. It _might_ just work for now by adjusting these values:
> 
> https://github.com/hankhank10/music-screen-api/blob/b29f2c0ec32b7da1fb0ab3515b249034379b8ff7/display_controller.py#L12-L15

Got the album art to adjust to the screen, there is a gray bar at the bottom when i use outside the standard parameters, not a big deal. One thing I did notice when the track title pops up is that it scales the image back outside the 480 width of the rectangle screen, Not sure if there is a way to adjust the text size down?

Everything else is working perfectly.

_Originally posted by @rdrivas123 in https://github.com/hankhank10/music-screen-api/pull/24#issuecomment-666066227_

jjlawren, 2020-07-30:
> SCREEN_W = 480
> SCREEN_H = 720
> THUMB_W = 600
> THUMB_H = 600

@rdrivas123 The above doesn't fit the screen size of the rectangular HyperPixel 4.0. If you're running in a vertical orientation, try these values:
```
SCREEN_W = 480
SCREEN_H = 800
THUMB_W = 480
THUMB_H = 480
```
It seems to roughly work but there might need some adjustments to the text in terms of placement, wrap width, etc.

rdrivas123, 2020-07-30:
> @rdrivas123 The above doesn't fit the screen size of the rectangular HyperPixel 4.0. If you're running in a vertical orientation, try these values:
>
> ```
> SCREEN_W = 480
> SCREEN_H = 800
> THUMB_W = 480 [...]

Made that adjustment looks good so far! Thanks!

craigusus, 2020-08-03:
Pulled new dev branch and looks pretty good to me!

```
SCREEN_W = 800
SCREEN_H = 480
THUMB_W = 350
THUMB_H = 350
```
and
```
if show_artist_and_album:
    track_font = tkFont.Font(family="Helvetica", size=25)
else:
    track_font = tkFont.Font(family="Helvetica", size=20)
detail_font = tkFont.Font(family="Helvetica", size=15)
```

[image]

vietscong, 2021-11-04:
> Pulled new dev branch and looks pretty good to me!
>
> ```
> SCREEN_W = 800
> SCREEN_H = 480
> THUMB_W = 350
> THUMB_H = 350
> ```
>
> and
>
> ```
> if show_artist_and_album:
>     track_font = [...]

Hi Craigusus

I have followed Mark Hank tutorial , and it´s working quitine fine , but is uses the rectangular hyperpixel screen ..and i wanted to use the adjustments values you uses too ..

I am stuck with my Pi because i cant get out of the Album art showing on the screen ..

How do you insert this new size value in the pi ?

Thanks for your help

Best regards

craigusus, 2021-11-04:
Hi @vietscong

The only files I edited were **_display_controller.py_** lines 13 - 16
from:
```
SCREEN_W = 720
SCREEN_H = 720
THUMB_W = 600
THUMB_H = 600
```
to
```
SCREEN_W = 800
SCREEN_H = 480
THUMB_W = 350
THUMB_H = 350
```
Lines 76 - 80
from:
```
if show_artist_and_album:
    track_font = tkFont.Font(family="Helvetica", size=30)
else:
    track_font = tkFont.Font(family="Helvetica", size=40)
detail_font = tkFont.Font(family="Helvetica", size=15)
```
to:
```
if show_artist_and_album:
    track_font = tkFont.Font(family="Helvetica", size=25)
else:
    track_font = tkFont.Font(family="Helvetica", size=20)
detail_font = tkFont.Font(family="Helvetica", size=15)
```

And **_sonos_settings.py_** (copy of my file below).
https://pastebin.com/nA5TyhRh

Please bear in mind I have my screen running on a separate Pi. So your value on line 6 of **_sonos_settings.py_** will probably still be:
```
sonos_http_api_address = "localhost"
```

I hope this helps.

vietscong, 2021-11-05:
Hi

Thank you very much for your time answering me.

I am just new to this and have a couple questions ..

Where and how i can acess and modify the file display_controller.py ?

Thanks

Best regards

https://pastebin.com/nA5TyhRh