hankhank10/music-screen-api #28: Getting artist/track/album info from streaming services (Inky)

yankeeatz, 2020-08-04:
I noticed on some streaming services Inky displays the full artist/track/album info, but on some it only shows the station name.  From my quick checks:

Spotify, Sonos Radio - show full info
Pandora, LivexLive, Radio Paradise - show only station name even though the full info is in the node-sonos-http-api info (and shows on the Sonos app)

Example from the Sonos API when playing Pandora:
{"volume":12,"mute":false,"equalizer":{"bass":0,"treble":0,"loudness":true},"currentTrack":{"artist":"The Wrights","title":"You Got The Thorns","album":"Down This Road","duration":242,"uri":"x-sonosapi-radio:ST%3a215077758810772746?sid=236&flags=8300&sn=1","trackUri":"x-sonos-http:PcKRzqA9dNH9sCmt1AR6InVkaCOWfpAdGEuIKqWlwdDiakrUDG871GeRUT5S3yfAz_F2kUb4Mzo-lgICUM79u352PxuY99Qty%3a%3aST%3a215077758810772746%3a%3aRINCON_000E58FAC84401400%3a240.mp3?sid=236&flags=0&sn=1","type":"radio","stationName":"The Wrights Radio (My Station)"},"nextTrack":{"artist":"","title":"","album":"","albumArtUri":"","duration":0,"uri":""},"trackNo":1,"elapsedTime":14,"elapsedTimeFormatted":"00:00:14","playbackState":"PLAYING","playMode":{"repeat":"none","shuffle":false,"crossfade":false}}

And Radio Paradise:
{"volume":12,"mute":false,"equalizer":{"bass":0,"treble":0,"loudness":true},"currentTrack":{"artist":"Grant Lee Buffalo","title":"Fuzzy","album":"Fuzzy","albumArtUri":"http://img.radioparadise.com/covers/l/B0000057BN.jpg","duration":290,"uri":"x-sonosapi-radio:0%3a10%3aresume?sid=308&flags=8224&sn=22","trackUri":"https://audio-geo.radioparadise.com/gapless/0/10/2027239.mp3/virtual/p/61336400/9223372036854775807/9223372036854775807/chan/0/10/2027232-2027239.mp3","type":"radio","stationName":"Main Mix","absoluteAlbumArtUri":"https://audio-geo.radioparadise.com/gapless/0/10/2027239.mp3/virtual/p/61336400/9223372036854775807/9223372036854775807/chan/0/10/2027232-2027239.mp3"},"nextTrack":{"artist":"Commercial-free","title":"Listener-supported","albumArtUri":"http://img.radioparadise.com/covers/l/103.jpg","duration":4,"uri":"https://audio-fb.radioparadise.com/blocks/chan/0/10/5007716.mp3","trackUri":"https://audio-fb.radioparadise.com/blocks/chan/0/10/5007716.mp3","absoluteAlbumArtUri":"https://audio-fb.radioparadise.com/blocks/chan/0/10/5007716.mp3"},"trackNo":1,"elapsedTime":184,"elapsedTimeFormatted":"00:03:04","playbackState":"TRANSITIONING","playMode":{"repeat":"none","shuffle":false,"crossfade":false}}

I gather I need to edit sonos_user_data.py in music-screen-api folder, probably in the section that defines how to handle the json data when radio is playing.  I have tried a few edits but none have gotten it working.

Can anyone else point me in the right direction?

jjlawren, 2020-08-04:
This is totally untested and you might get mixed results, but you can just skip the radio-specific handling. Here's a suggestion to get you started: https://github.com/jjlawren/music-screen-api/blob/disable_radio_hack/sonos_user_data.py

yankeeatz, 2020-08-04:
Thanks, that was what I tried already and it didn't work.  I re-tried with your code just in case I missed anything and still no go.

jjlawren, 2020-08-04:
If it's still displaying the station name then I'd double check that you installed it correctly. With the `sonos_user_data.py` file linked above that shouldn't be possible anymore.

yankeeatz, 2020-08-04:
I just tried again and yes, it's working.  Don't know what I did differently this time (copy new file over old and reboot) but nevertheless, it's working.   Thanks again for the fast response - closing!

closed 2020-08-04 by yankeeatz