hankhank10/music-screen-api #29: Dev wont launch

rdrivas123, 2020-08-05:
Weird error:

I got a new 4.0 screen, set it up using the standard setup and then did a git pull
It tells me I am up to date, I change to dev (git checkout dev) then run git pull again, it tells me everything is up to date (I figured i would see new files come down)

Now when i reboot I cannot get the app to launch, I have to run:   DISPLAY=:0 /usr/bin/python3 /home/pi/music-screen-api/go_sonos_highres.py

This works for me, but once i CTRL+C in the SSH window or if i disconnect, it no longer displays album art.
I cant for the life of me figure out why it wont select the display.

When running it without the display command I get the following:

2020-08-04 21:51:55,820   ERROR - Task exception was never retrieved
future: <Task finished coro=<main() done, defined at music-screen-api/go_sonos_highres.py:144> exception=TclError('no display name                                                                      and no $DISPLAY environment variable')>
Traceback (most recent call last):
  File "music-screen-api/go_sonos_highres.py", line 148, in main
    display = DisplayController(loop, sonos_settings.show_details, sonos_settings.show_artist_and_album, show_details_timeout)
  File "/home/pi/music-screen-api/display_controller.py", line 35, in __init__
    self.root = tk.Tk()
  File "/usr/lib/python3.7/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable

jjlawren, 2020-08-05:
If you're running from an SSH connection you need to set the `DISPLAY` environment variable. This is not necessary if you're connected using VNC and open a terminal window from within the GUI.

Does autostart work properly on reboot?

rdrivas123, 2020-08-05:
No, no autostart on reboot.

jjlawren, 2020-08-05:
Check the script logs and also `~/.xsession-errors`.

rdrivas123, 2020-08-05:
No error in xsession, still get this in the app log:

future: <Task finished coro=<main() done, defined at /home/pi/music-screen-api/go_sonos_highres.py:144> exception=TclError('no display name and no $DISPLAY environment variable')>
Traceback (most recent call last):
  File "/home/pi/music-screen-api/go_sonos_highres.py", line 148, in main
    display = DisplayController(loop, sonos_settings.show_details, sonos_settings.show_artist_and_album, show_details_timeout)
  File "/home/pi/music-screen-api/display_controller.py", line 35, in __init__
    self.root = tk.Tk()
  File "/usr/lib/python3.7/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable
2020-08-04 22:08:08,208   ERROR - Task exception was never retrieved
future: <Task finished coro=<main() done, defined at /home/pi/music-screen-api/go_sonos_highres.py:144> exception=TclError('no display name and no $DISPLAY environment variable')>
Traceback (most recent call last):
  File "/home/pi/music-screen-api/go_sonos_highres.py", line 148, in main
    display = DisplayController(loop, sonos_settings.show_details, sonos_settings.show_artist_and_album, show_details_timeout)
  File "/home/pi/music-screen-api/display_controller.py", line 35, in __init__
    self.root = tk.Tk()
  File "/usr/lib/python3.7/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable

jjlawren, 2020-08-05:
This is a general error saying that it's being launched from outside of the GUI. What does your `/etc/xdg/lxsession/LXDE-pi/autostart` file look like?

rdrivas123, 2020-08-05:
GNU nano 3.2                                                                  /etc/xdg/lxsession/LXDE-pi/autostart

@lxpanel --profile LXDE-pi
@pcmanfm --desktop --profile LXDE-pi
@xscreensaver -no-splash
@xset s noblank
@xset s off
@xset -dpms
@sudo /usr/bin/python3 /home/pi/music-screen-api/go_sonos_highres.py

jjlawren, 2020-08-05:
Can you try to change the last line to:
```
@/usr/bin/python3 /home/pi/music-screen-api/go_sonos_highres.py
```
It may complain about missing dependencies after that, but this script shouldn't need root permissions.

rdrivas123, 2020-08-05:
No dependencies issues, its working now. As I build more units I will check the info you sent.

closed 2020-08-05 by jjlawren