hankhank10/music-screen-api #30: feature request: turn screen off when source is line_in

habra-cadabra, 2020-08-06:
hello!

I've tried poking around to add this, but I think I'm in way over my head. 

I'm using this setup in my family room, and when we watch TV at night the display shows the album art of the last album we were listening to all night and at peak brightness. it would be preferable if there was an option to turn the screen off when the source is detected as "line_in". 

The state of my system when watching TV looks like this (I crossed out the RINCON value, not sure what that is exactly):

`{
"volume":21,
"mute":false,
"equalizer":
{
"bass":0,
"treble":0,
"loudness":true,
"speechEnhancement":true,
"nightMode":false
},
"currentTrack":
{
"duration":0,
"uri":"x-sonos-htastream:RINCON_XXXXXXXXXXXXXXXXXX:spdif",
"trackUri":"x-sonos-htastream:RINCON_XXXXXXXXXXXXXXXXXX:spdif",
"type":"line_in",
"stationName":""
}
"nextTrack":
{
"artist":"",
"title":"",
"album":"",
"albumArtUri":"",
"duration":0,
"uri":""
},
"trackNo":1,
"elapsedTime":4101,
"elapsedTimeFormatted":"01:08:21",
"playbackState":"PLAYING",
"playMode":
{
"repeat":"none",
"shuffle":false,
"crossfade":false
},
"sub":
{
"gain":6,
"crossover":0,
"polarity":0,
"enabled":true
}
}`

jjlawren, 2020-08-06:
Should be straightforward. Is there ever a time the screen shows something useful when using line-in?

habra-cadabra, 2020-08-06:
no, not really! the screen shows the album art of the last played track the entire time.

I think it could be cool to detect what's being played via line in and have the screen display a little TV icon or a line-in icon. 

from browsing the node-sonos-http documentation, it looks like home theater sonos products will show something like the following in the "uri" field when playing from hdmi/optical: "x-sonos-htastream:RINCON_XXXXXXXXXXXXX01400:spdif"

but the sonos play:5/five will display "x-sonos-stream:RINCON_XXXXXXXXXXXXX01400" when the source is the line-in on the speaker. 

[image]

[image]

closed 2020-08-13 by jjlawren