hankhank10/music-screen-api #31: AttributeError: 'SonosData' object has no attribute 'type'

davidphillipsuk, 2020-08-08:
I'm getting an error with the most recent version of the display_controller.py file:

2020-08-08 12:04:36,349   ERROR - Error handling request
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/aiohttp/web_protocol.py", line 418, in start
    resp = await task
  File "/home/pi/.local/lib/python3.7/site-packages/aiohttp/web_app.py", line 458, in _handle
    resp = await handler(request)
  File "/home/pi/music-screen-api/webhook_handler.py", line 68, in handle_webhook
    await self.callback()
  File "/home/pi/music-screen-api/go_sonos_highres.py", line 171, in webhook_callback
    await redraw(session, sonos_data, display)
  File "/home/pi/music-screen-api/go_sonos_highres.py", line 103, in redraw
    display.update(pil_image, sonos_data)
  File "/home/pi/music-screen-api/display_controller.py", line 159, in update
    if sonos_data.type == "radio":
AttributeError: 'SonosData' object has no attribute 'type'

This is when I'm playing a Spotify playlist. It seems to occasionally prevent the album art displaying on the screen, or has a delay in doing so. 

Thanks for all the work you put into this - it's awesome!

davidphillipsuk, 2020-08-10:
All sorted - looks like I was missing the updates in sonos_user_data.py.

closed 2020-08-10 by davidphillipsuk

jjlawren, 2020-08-10:
Can you elaborate? Did you have to make any changes?

davidphillipsuk, 2020-08-10:
I thought I'd pulled all of the changes to the dev branch, but it appears there were outstanding changes to sonos_user_data.py when I pulled again. It's working perfectly again now. Thanks!

jjlawren, 2020-08-10:
BTW, all of the dev changes have been merged into the master branch. You'll probably want to switch back when you have time to get future updates.