hankhank10/music-screen-api #36: no longer autostarts after reboot

habra-cadabra, 2020-08-14:
hello!

music-screen-api doesn't launch at startup starting at commit #33. I reverted back to #32 and everything starts as normal. 

running "DISPLAY=:0 /usr/bin/python3 /home/pi/music-screen-api/go_sonos_highres.py" on #33 or #34 starts music-screen-api normally.

jjlawren, 2020-08-14:
Any logs or tracebacks?

habra-cadabra, 2020-08-14:
yep, sorry! got sidetracked. 

2020-08-13 22:09:56,112   ERROR - Task exception was never retrieved
future: <Task finished coro=<main() done, defined at /home/pi/music-screen-api/go_sonos_highres.py:171> exception=CalledProcessError(128, ['git', 'describe'])>
Traceback (most recent call last):
  File "/home/pi/music-screen-api/go_sonos_highres.py", line 174, in main
    log_git_hash()
  File "/home/pi/music-screen-api/go_sonos_highres.py", line 129, in log_git_hash
    git_hash = subprocess.check_output(["git", "describe"], text=True).strip()
  File "/usr/lib/python3.7/subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.7/subprocess.py", line 487, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['git', 'describe']' returned non-zero exit status 128.

jjlawren, 2020-08-14:
Ok thanks, the reason is clear. I'll try to take a look at it tomorrow.

PR #37 (2020-08-14): Fix failing autostart
Subprocess command assumed the script would run from the repo directory. This explicitly sets the cwd and catches another exception as a fallback.

Fixes: #36
commit 84f550b9 (2020-08-14): Fix failing autostart (#37)

Subprocess command assumed the script would run from the repo directory. This explitly sets the cwd and catches another exception as a fallback.

Fixes: #36
closed 2020-08-14 by jjlawren