hankhank10/music-screen-api #44: Screen not updating - demaster offline?

yankeeatz, 2020-10-12:
Noticed my Inky not updating this morning.  Trying to troubleshoot.  I confirmed the Sonos server API (on a PI 3) is running fine, and the Pi Zero running music-screen-api can see it.    Rebooted both Pi's, still no go.

I tried manually starting music-screen-api and get the following; am I reading correctly that the demaster server is offline and that's what is breaking the display?

pi@sonospizero:~ $ python3 music-screen-api/go_sonos.py
Enter a Sonos room name >>>  Office
Checking API for last played: new data found from api - refreshing screen
Checking API at http://demaster.hankapi.com/demaster?format=simple&long_track_na                                                                             me=Heartache Tonight
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 384, in                                                                              _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 380, in                                                                              _make_request
    httplib_response = conn.getresponse()
  File "/usr/lib/python3.7/http/client.py", line 1336, in getresponse
    response.begin()
  File "/usr/lib/python3.7/http/client.py", line 306, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.7/http/client.py", line 267, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/usr/lib/python3.7/socket.py", line 589, in readinto
    return self._sock.recv_into(b)
  File "/usr/lib/python3.7/ssl.py", line 1052, in recv_into
    return self.read(nbytes, buffer)
  File "/usr/lib/python3.7/ssl.py", line 911, in read
    return self._sslobj.read(len, buffer)
socket.timeout: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 638, in                                                                              urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 367, in incr                                                                             ement
    raise six.reraise(type(error), error, _stacktrace)
  File "/usr/lib/python3/dist-packages/six.py", line 693, in reraise
    raise value
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in                                                                              urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 386, in                                                                              _make_request
    self._raise_timeout(err=e, url=url, timeout_value=read_timeout)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 306, in                                                                              _raise_timeout
    raise ReadTimeoutError(self, url, "Read timed out. (read timeout=%s)" % time                                                                             out_value)
urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='demaster.hankapi.                                                                             com', port=443): Read timed out. (read timeout=5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/music-screen-api/demaster.py", line 50, in strip_name_api
    r = requests.get(api_url, timeout=5)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 533, in reque                                                                             st
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 668, in send
    history = [resp for resp in gen] if allow_redirects else []
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 668, in <list                                                                             comp>
    history = [resp for resp in gen] if allow_redirects else []
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 247, in resol                                                                             ve_redirects
    **adapter_kwargs
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 529, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='demaster.hankapi.com'                                                                             , port=443): Read timed out. (read timeout=5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "music-screen-api/go_sonos.py", line 62, in <module>
    current_track = demaster.strip_name (current_track)
  File "/home/pi/music-screen-api/demaster.py", line 73, in strip_name
    online_api_response = strip_name_api(full_song_name)
  File "/home/pi/music-screen-api/demaster.py", line 52, in strip_name_api
    print ("Error contacting demaster host: " + err)
TypeError: can only concatenate str (not "ReadTimeout") to str

habra-cadabra, 2020-10-15:
Hello! I’ve been experiencing issues the past several days related to screen responsiveness.

 I saw this comment, and changed demaster_query_cloud to false in the sonos_settings.py file, and looks like everything is behaving as expected now.

sktusing, 2020-11-16:
Seeing the same issue however changing demaster_query_cloud in sonos_settings.py does not solve the issue.

`pi@raspberrypi:~ $ python3 music-screen-api/go_sonos.py
Enter a Sonos room name >>>  Office
Pausing for 60 seconds on startup to let pi zero catch up
Checking API for last played: new data found from api - refreshing screen
Checking API at http://demaster.hankapi.com/demaster?format=simple&long_track_name=Another One Bites The Dust - Remastered 2011
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 384, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 380, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/lib/python3.7/http/client.py", line 1336, in getresponse
    response.begin()
  File "/usr/lib/python3.7/http/client.py", line 306, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.7/http/client.py", line 267, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/usr/lib/python3.7/socket.py", line 589, in readinto
    return self._sock.recv_into(b)
  File "/usr/lib/python3/dist-packages/urllib3/contrib/pyopenssl.py", line 307, in recv_into
    raise timeout('The read operation timed out')
socket.timeout: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 367, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/usr/lib/python3/dist-packages/six.py", line 693, in reraise
    raise value
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 386, in _make_request
    self._raise_timeout(err=e, url=url, timeout_value=read_timeout)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 306, in _raise_timeout
    raise ReadTimeoutError(self, url, "Read timed out. (read timeout=%s)" % timeout_value)
urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='demaster.hankapi.com', port=443): Read timed out. (read timeout=5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/music-screen-api/demaster.py", line 50, in strip_name_api
    r = requests.get(api_url, timeout=5)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 668, in send
    history = [resp for resp in gen] if allow_redirects else []
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 668, in <listcomp>
    history = [resp for resp in gen] if allow_redirects else []
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 247, in resolve_redirects
    **adapter_kwargs
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 529, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='demaster.hankapi.com', port=443): Read timed out. (read timeout=5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "music-screen-api/go_sonos.py", line 62, in <module>
    current_track = demaster.strip_name (current_track)
  File "/home/pi/music-screen-api/demaster.py", line 73, in strip_name
    online_api_response = strip_name_api(full_song_name)
  File "/home/pi/music-screen-api/demaster.py", line 52, in strip_name_api
    print ("Error contacting demaster host: " + err)
TypeError: can only concatenate str (not "ReadTimeout") to str
`

sktusing, 2020-11-16:
However, changing "demaster" to False in sonos_settings.py **does** solve the issue

jjlawren, 2020-11-16:
Yes, the online demastering service seems unavailable. @sktusing, it looks like you're using the older e-ink script which has not been updated for offline demastering. You'll need to disable it completely for now.

hankhank10, 2020-11-16:
Sorry, looks like there was an error on the server end. Should be fixed now.

closed 2023-01-05 by hankhank10