hankhank10/music-screen-api #46: show_artist_and_album not behaving as expected

habra-cadabra, 2020-10-15:
Hello! 

I’ve Noticed that the show_artist_and_album setting doesn’t seem to behave according to the description. 

When set to true the artist and album are displayed as expected. When set to false album title is still displayed. 

Perhaps this setting could have 4 values? Off, artist only, album only, album & artist?

This is an example set to false: 

[image]

hankhank10, 2020-10-15:
There are two settings:

`show_details` determines whether there is any detail shown at all. False shows just the album art, True shows some details.

The level of details displayed is determined by `show_artist_and_album`

I thought this was clear from the comment (Ignored if 'show_details' is False) but very open to ideas on how this could be clearer.

jjlawren, 2020-10-15:
I thought it should only display the track name (not the album) with the settings you're using. 

What version of the script are you running and what's the media source?

habra-cadabra, 2020-10-18:
ok, I was able to go back through some builds and pinpoint when the behavior changed. 

this is build b69d862 form Aug. 1, behaving as expected:

[image]

note the album or artist is not showing, however, there seems to be a weird glitch in the text in the 2nd line.

this is build 8861bfe from Aug. 3, when the behavior of the show_artist_and_album changed:

[image]

note the album is showing, but not the artist.

jjlawren, 2020-10-18:
Thanks for putting in the effort to track this down. Looks like it's probably related to my changes. I'll take a look as soon as I can.

jjlawren, 2020-11-10:
@habra-cadabra I've created a PR to fix this. I don't remember why some of the checks existed in the code I replaced. Would you be able to validate a bit on that branch before we merge?

habra-cadabra, 2020-11-11:
nice! I'll try and check this in the next day or so. thanks!

habra-cadabra, 2020-11-15:
hey there, sorry for the delay!

running build c2aa8e4, and it looks good and works as expected. Thanks for sorting that out! much appreciated. 

[image]

commit 205ff52a (2020-11-15): Fix show_artist_and_album behavior (#49)

Fixes #46
closed 2020-11-15 by jjlawren