hankhank10/music-screen-api #47: Fresh install not working go_sonos_highres.py crashes

ashenshugarRET, 2020-10-24:
I have performed a new install of the music-screen-api and have got to step 7 of the install instruction and am getting th efollowing error:

pi@raspberrypi:~ $ python3 music-screen-api/go_sonos_highres.py
2020-10-24 11:34:01,515    INFO - Writing to log file: /home/pi/music-screen-api.log
2020-10-24 11:34:01,533    INFO - Current script version: v0.4.0-78-g9e2f533
2020-10-24 11:34:01,543   ERROR - Task exception was never retrieved
future: <Task finished coro=<main() done, defined at music-screen-api/go_sonos_highres.py:152> exception=TclError('no display name and no $DISPLAY environment variable')>
Traceback (most recent call last):
  File "music-screen-api/go_sonos_highres.py", line 157, in main
    display = DisplayController(loop, sonos_settings.show_details, sonos_settings.show_artist_and_album, show_details_timeout)
  File "/home/pi/music-screen-api/display_controller.py", line 35, in __init__
    self.root = tk.Tk()
  File "/usr/lib/python3.7/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable

jjlawren, 2020-10-24:
If you're connecting over SSH you'll need to add another environment variable to the command to access the display like this:

`pi@raspberrypi:~ $ DISPLAY=:0 python3 music-screen-api/go_sonos_highres.py`

closed 2020-10-24 by ashenshugarRET

jjlawren, 2020-10-24:
I submitted a PR to handle this scenario a bit better.