hankhank10/music-screen-api #50: Albumart not showing on screen connect to Raspberry Pi running node-sonos-http-api if music started from Spotify app

ashenshugarRET, 2020-11-15:
My screen which is connected directly to the Raspberry Pi running my install of node-sonos-http-api doesn't show album art if the music is started from the spotify app:
[image]

However, it does show from my second screen which listens to the IP address of the Pi with node-sonos-http-api installed
[image]

The album art shows correctly on both screens if the music is started either by vinylemulator NFC tags (again connected to the Pi running  node-sonos-http-api) or through the SONOS phone app.

Any thoughts on what is wrong?

jjlawren, 2020-11-15:
What version is each instance using? Is either using webhooks to update? Can you share logs?

ashenshugarRET, 2020-11-15:
All instances are the most up to date from the git repositories for vinylemulator, node-sonos-http-api and music-screen-api.
I've configured node-sonos-http-api to use webhooks by creating the settings.json file as suggested. Do music-screen-api also need to be configured to use webhooks?
I can't find any logs that are updating.

jjlawren, 2020-11-15:
Nothing to configure for webhooks on this side, but we can tell if it's working from the logs. Make sure these are in your `sonos_settings.py` file(s):
```
log_file = "~/music-screen-api.log"
log_level = "DEBUG"
```
The logfile will be written to the user's home directory, which will likely be `/home/pi/music-screen-api.log`.

ashenshugarRET, 2020-11-15:
Thanks for your help.
I have both of those lines in the sons_settings.py files for both installations of music-screen-api, however nothing is being written to them. I have the installs configured to automatically start following the hackster.io tutorial, could that cause a problem with logging correctly?

jjlawren, 2020-11-15:
No, I would still expect it to write to the log. Can you run `git describe` in each checkout? You can also run the script over SSH and it will print log lines directly to the console.

ashenshugarRET, 2020-11-15:
I've managed to get the log printing, the log for the screen which doesn't work when I start a song from the Spotify app is:

2020-11-15 15:52:55,848    INFO - New track: Lady Gaga - Stupid Love (Chromatica) - 0:03:14
2020-11-15 15:52:55,943   ERROR - SSL handshake failed on verifying the certificate
protocol: <asyncio.sslproto.SSLProtocol object at 0x743947f0>
transport: <_SelectorSocketTransport fd=13 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/sslproto.py", line 625, in _on_handshake_complete
    raise handshake_exc
  File "/usr/lib/python3.7/asyncio/sslproto.py", line 189, in feed_ssldata
    self._sslobj.do_handshake()
  File "/usr/lib/python3.7/ssl.py", line 763, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: IP address mismatch, certificate is not valid for '199.232.54.248'. (_ssl.c:1056)
2020-11-15 15:52:55,945   ERROR - SSL error in data received
protocol: <asyncio.sslproto.SSLProtocol object at 0x743947f0>
transport: <_SelectorSocketTransport closing fd=13 read=idle write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/sslproto.py", line 526, in data_received
    ssldata, appdata = self._sslpipe.feed_ssldata(data)
  File "/usr/lib/python3.7/asyncio/sslproto.py", line 189, in feed_ssldata
    self._sslobj.do_handshake()
  File "/usr/lib/python3.7/ssl.py", line 763, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: IP address mismatch, certificate is not valid for '199.232.54.248'. (_ssl.c:1056)
2020-11-15 15:52:55,948 WARNING - Problem connecting to https://i.scdn.co/image/ab67616d0000b2736040effba89b9b00a6f6743a [Cannot connect to host i.scdn.co:443 ssl:default [Network is unreachable]]
2020-11-15 15:52:55,949 WARNING - Image not available, using default

ashenshugarRET, 2020-11-15:
git describe information:
Master Raspberry Pi
music-screen-api: v0.4.0-80-g205ff52
node-sonos-http-api: v1.4.3-52-g7b5d5e4
vinylemulator:  v0.9.3

Second music-screen Pi
music-screen-api: v0.4.0-79-g7c575f8

I now have a log output for the second screen, whilst everything seems to be working, I'm getting a lot of aiohttp server disconnect errors:

2020-11-15 16:37:36,769    INFO - New track: Kygo - Higher Love (Higher Love) - 0:03:48
2020-11-15 16:39:25,981   ERROR - Error connecting to Sonos API: [Errno 104] Connection reset by peer
Traceback (most recent call last):
  File "/home/pi/music-screen-api/sonos_user_data.py", line 153, in refresh
    async with self.session.get(url) as response:
  File "/usr/local/lib/python3.7/dist-packages/aiohttp/client.py", line 1124, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.7/dist-packages/aiohttp/client.py", line 551, in _request
    await resp.start(conn)
  File "/usr/local/lib/python3.7/dist-packages/aiohttp/client_reqrep.py", line 890, in start
    message, payload = await self._protocol.read()  # type: ignore  # noqa
  File "/usr/local/lib/python3.7/dist-packages/aiohttp/streams.py", line 605, in read
    await self._waiter
aiohttp.client_exceptions.ClientOSError: [Errno 104] Connection reset by peer
2020-11-15 16:40:01,310   ERROR - Error connecting to Sonos API: Server disconnected
Traceback (most recent call last):
  File "/home/pi/music-screen-api/sonos_user_data.py", line 153, in refresh
    async with self.session.get(url) as response:
  File "/usr/local/lib/python3.7/dist-packages/aiohttp/client.py", line 1124, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.7/dist-packages/aiohttp/client.py", line 551, in _request
    await resp.start(conn)
  File "/usr/local/lib/python3.7/dist-packages/aiohttp/client_reqrep.py", line 890, in start
    message, payload = await self._protocol.read()  # type: ignore  # noqa
  File "/usr/local/lib/python3.7/dist-packages/aiohttp/streams.py", line 605, in read
    await self._waiter
aiohttp.client_exceptions.ServerDisconnectedError: Server disconnected
2020-11-15 16:41:24,130    INFO - New track: Lady Gaga - Stupid Love (Chromatica) - 0:03:13
2020-11-15 16:44:37,691    INFO - New track: Jonas Blue - What I Like About You (feat. Theresa Rex) (What I Like About You (feat. Theresa Rex)) - 0:03:40
2020-11-15 16:44:37,692   DEBUG - Demastered to What I Like About You

jjlawren, 2020-11-15:
The issue on your primary is SSL related, which could be related to an outdated CA store, bad DNS, or other issues. As a sanity check, run `sudo update-ca-certificates` on that machine to make sure the CA certs are installed/current.

The secondary host errors look like networking issues, or the `node-sonos-http-api` service is having problems or is getting overloaded. What hardware are each of these running on?

ashenshugarRET, 2020-11-15:
Thanks, 

I've run  `sudo update-ca-certificates` and get this output:
```
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
```
So I think that is OK,

both Raspberry Pi's are 3A.

As the issue is at my end I'll close the

closed 2020-11-15 by ashenshugarRET