hankhank10/music-screen-api #55: Title track name text is over written by artist/album text when the track name is too long for a single line

ashenshugarRET, 2022-02-10:
Love this app, however, I've noticed that when the music I'm listening to has a very long title the artist/album text overwrites the second line of the title
[image]

I can adapt the display_controller.py file to reduce the thumbnail size by updating lines 12-16 in the standard file to the following:
`SCREEN_W = 720
SCREEN_H = 720
THUMB_W = 580
THUMB_H = 580`

and to shrink the text by using by updating lines  73-77:
`        if show_artist_and_album:
            track_font = tkFont.Font(family="Helvetica", size=30)
        else:
            track_font = tkFont.Font(family="Helvetica", size=40)
        detail_font = tkFont.Font(family="Helvetica", size=15)`

My question is can the code be updated to respond to the length of the track name being displayed and use these updated settings only if the track name is going to be displayed on two lines?

@jjlawren As you wrote the original code, can you help?

ashenshugarRET, 2022-02-10:
I've managed to implement this in pull request #57
[image]

ashenshugarRET, 2022-02-11:
Here is a better display of the improved handling of track title, album and artist text lengths.
[image]

justme-1968, 2022-02-11:
as you already had a look at the source...

i just opened an issue regarding the idea that it would be better to habe the text overlay the image instead of scaling the image. that would avoid flickering and also the image would habe to be scaled only once.

maybe you already have seen how this could be implemented?

ashenshugarRET, 2022-02-11:
> as you already had a look at the source...
> 
> i just opened an issue regarding the idea that it would be better to habe the text overlay the image instead of scaling the image. that would avoid flickering and also the image would habe to be scaled only once.
> 
> this in combination with your changes would be great.
> 
> maybe you already have seen how this could be implemented?

Hello,

Yes, I saw this suggestion, I'm not sure I have the skill to implement it though, as my solution only required minor modification of @jjlawren code. I'll have a try as I like a challenge.

ashenshugarRET, 2022-02-12:
@justme-1968 is this the effect you are looking for?
[image]

justme-1968, 2022-02-12:
yes! exactly

ashenshugarRET, 2022-02-12:
OK, great. I'm going to close my current pull requests and create a single one with the improvement to text handling and BBC Sounds service in a single request.

ashenshugarRET, 2022-02-12:
This is fixed in pull request #58

justme-1968, 2022-02-12:
great! thank you.