hankhank10/music-screen-api #67: Feature Request: Specify font family in settings

tonyrobots, 2024-02-07:
Hi, would be great to be able to specify the font family for details in the sonos_settings.py file. I might take a swing at this myself, but wanted to post in case there's any direction or gotchas worth knowing about. Looks like some of the infrastructure to support this was put in place but just never completed. 

Thanks!

ashenshugarRET, 2024-02-07:
I did some work on the overlay feature to show artist and track names, a big gotcha of being able to choose the font is that most fonts have letters of different widths making predicting how the text will fit on the screen almost impossible. In the end, I went for the monospaced font Consolas as I liked the look of it and it behaved predictably when calculating the length of the text strings for resizing etc...

ashenshugarRET, 2024-02-07:
I have a working version of this functionality on my repository forked from this one, I'm going to test it for a few days so I can be sure it isn't doing anything unexpected and if all goes well I'll raise a pull request to incorporate it into this repository.

tonyrobots, 2024-02-07:
Right, that makes sense. I see there's some hardcoded bits in there like if str_length > 26, use font_size = 14, else use font_size = 18, and that sort of thing, which can be tricky with a variable width font. But still it would be good to give users the option, I think. If you don't mind sharing your forked repo I'd love to give it a spin.

ashenshugarRET, 2024-02-08:
Hello no problem my repository is here [https://github.com/ashenshugarRET/music-screen-api.git](https://github.com/ashenshugarRET/music-screen-api.git)