hankhank10/music-screen-api #68: Question/suggestion about the sonos Node API dependency

mattboldt, 2024-02-12:
First off, thank you for your blog posts and this library! I found it very easy to get set up and running with very limited experience with raspberry pi and python.

I'm curious about the choice of using `node-sonos-http-api` instead of something in python, like https://github.com/SoCo/SoCo. My understanding is that by using the python library, you won't have to run an http server on the Pi, which seems less resource intensive. That, and you get to stay in one language.

I'm only looking to display basic information like play status and volume on an OLED, so I'm gonna try swapping in SoCo, but I wanted to ask if you tried it before the node server.

jjlawren, 2024-02-12:
I've wanted to do exactly this but haven't had the spare time. It's definitely doable, but would not be a simple drop-in swap.

mattboldt, 2024-02-12:
If my OLED project works out, I can submit a PR in case it's a helpful starting point.

hankhank10, 2024-02-12:
Thanks @mattboldt and glad you are enjoying this.

When I built this in 2020 it was my first python project and first Pi project (and my first programming project!) so I tried to get to a "minimum viable product" relying as much as possible on pre-existing tools. At the time the Node Sonos HTTP library was well supported and seemed an easy (albeit maybe not elegant) plug and play solution so I went with that.

I agree that today if there's an "all python" version of this then that would be an improvement, and I would be really pleased to see you fork this repo and move it forward in that direction.